robocup-junior/rcj-soccersim #5: JSON-based game logs

mrshu, 2020-11-23:
We'll need to have logs on two levels:

1. Debug logs (probably via Python's `logging`)
2. Game logs, which ought to be readable outside of `webots` as well as Python itself

I think one of the better options would be having some JSON lines (https://jsonlines.org/) or something of that sort with action that happened (goal scored, lack of progress, robot's position restarted, ball's position restarted....) as well as some metadata, chiefly among them the timestamp at which the event took place. This would allow us to not just have historical data relevant matches but also to better navigate their recordings (i.e. it should be trivial to see all timestamps at which lack-of-progress took place for instance).

@felipenmartins @javier687 @marcodankel @Adman what do you think

Adman, 2020-11-24:
Definitely agree.

This could also lead to getting some statistics of each match / entire tournament.

felipenmartins, 2020-11-25:
I totally agree with you!

Felipe Nascimento Martins
felipe.n.martins@gmail.com
http://about.me/felipe.n.martins

dbscoach, 2020-12-07:
Positions of all Objects in the game can also be extracted from the websocket for the x3d viewer.

dbscoach, 2020-12-21:
I have a quick-and-dirty python script that captures the whole game (except maybe the first couple hundred milliseconds, depending on startup sequencing) in #25

mrshu, 2020-12-21:
Thanks @dbscoach, that's quite useful!

What I had in mind was a set of JSON logs that would also describe various game-specific actions (i.e. goal scored, lack of progress and so on) and will probably us to change the Supervisor (such that it would emit these events somewhere).

So I guess we'll have three set of JSON-based game logs:

1. Low level "where is this object positioned" logs created by #25 
2. Debug logs (probably via Python's logging)
3. Game logs, which ought to be readable outside of webots as well as Python itself

It's coming along rather nicely!

dbscoach, 2020-12-21:
I see. I have only taken a brief look at the supervisor but will try to make myself useful there. 
The x3d does capture the stdout and stderr of webots during the run (configurable, but I'd rather grep it out in post that not capture) so for a quick and dirty solution just print logging info to stdout and we can grep them out of the x3d-log.

mrshu, 2020-12-21:
I'll try to put together the wrapper real quick, so that there will be a separate file (i.e. not stdout or stderr) to which the "reflog" will be appended. Don't worry about grepping it out just yet :)

commit a60295b9 (2020-12-21): add: JSON logging

* Add a way of logging events that take place in the referre
  (JSONLogger), which has no dependences outside of Python standard
  library that sends JSON lines to a file.

* Introduce the concept of team names (so that the Supervisor can make
  use of them, in say the logs).

* Put together a way of creating a filename (the `reflog_path` function)

* Fixes #5.
PR #26 (2020-12-29): add: JSON logging
* Add a way of logging events that take place in the referre
  (JSONLogger), which has no dependences outside of Python standard
  library that sends JSON lines to a file.

* Introduce the concept of team names (so that the Supervisor can make
  use of them, in say the logs).

* Put together a way of creating a filename (the `reflog_path` function)

* Fixes #5.
closed 2020-12-29 by mrshu