robocup-junior/rcj-soccersim #17: Auto Gameplay Pipeline

aquahika, 2020-12-07:
Pack it in docker?
Input several student's .py files and output game playback and statistics.

aquahika, 2020-12-16:
Running a webots on Docker is introduced at the official website:
https://www.cyberbotics.com/doc/guide/installation-procedure#installing-the-docker-image

On my environment with nvidia GPU, after installing nvidia-docker2 package it works well: 
[image]

dbscoach, 2020-12-21:
Is there any easy way to run this headless without losing GPU acceleration? I tried and failed...

mrshu, 2021-01-02:
@dbscoach @aquahika We should probably get together and figure out whether we can actually use GPU acceleration. I am happy to provide a server for testing purposes as well, if that would be of any help.

Would something like that work for you?

Thanks!

closed 2021-03-15 by Adman