robocup-junior/rcj-soccersim #19: Robot B not moving when running simulation

jamestalkington, 2020-12-13:
OS: Windows 10 build 19041.685
Python: 3.7.9
Webots: R2020b Revision 2
Error:
```INFO: rcj_soccer_player: Starting controller: python3.exe -u rcj_soccer_player.py
INFO: rcj_soccer_player: Starting controller: python3.exe -u rcj_soccer_player.py
INFO: rcj_soccer_player: Starting controller: python3.exe -u rcj_soccer_player.py
INFO: rcj_soccer_player: Starting controller: python3.exe -u rcj_soccer_player.py
INFO: rcj_soccer_player: Starting controller: python3.exe -u rcj_soccer_player.py
INFO: rcj_soccer_player: Starting controller: python3.exe -u rcj_soccer_player.py
INFO: rcj_soccer_referee_supervisor: Starting controller: python3.exe -u rcj_soccer_referee_supervisor.py
Traceback (most recent call last):
  File "rcj_soccer_player.py", line 2, in <module>
    from controller import Robot
  File "C:\Program Files\Webots\lib\controller\python37\controller.py", line 31, in <module>
    import _controller
ImportError: DLL load failed: The specified module could not be found.
WARNING: 'rcj_soccer_player' controller exited with status: 1.
Traceback (most recent call last):
  File "rcj_soccer_player.py", line 2, in <module>
    from controller import Robot
  File "C:\Program Files\Webots\lib\controller\python37\controller.py", line 31, in <module>
    import _controller
ImportError: DLL load failed: The specified module could not be found.
WARNING: 'rcj_soccer_player' controller exited with status: 1.
Traceback (most recent call last):
  File "rcj_soccer_player.py", line 2, in <module>
    from controller import Robot
  File "C:\Program Files\Webots\lib\controller\python37\controller.py", line 31, in <module>
    import _controller
ImportError: DLL load failed: The specified module could not be found.
WARNING: 'rcj_soccer_player' controller exited with status: 1.
Traceback (most recent call last):
  File "rcj_soccer_player.py", line 2, in <module>
    from controller import Robot
  File "C:\Program Files\Webots\lib\controller\python37\controller.py", line 31, in <module>
    import _controller
ImportError: DLL load failed: The specified module could not be found.
WARNING: 'rcj_soccer_player' controller exited with status: 1.
Traceback (most recent call last):
  File "rcj_soccer_player.py", line 2, in <module>
    from controller import Robot
  File "C:\Program Files\Webots\lib\controller\python37\controller.py", line 31, in <module>
    import _controller
ImportError: DLL load failed: The specified module could not be found.
WARNING: 'rcj_soccer_player' controller exited with status: 1.
Traceback (most recent call last):
  File "rcj_soccer_player.py", line 2, in <module>
    from controller import Robot
  File "C:\Program Files\Webots\lib\controller\python37\controller.py", line 31, in <module>
    import _controller
ImportError: DLL load failed: The specified module could not be found.
WARNING: 'rcj_soccer_player' controller exited with status: 1.
Traceback (most recent call last):
  File "rcj_soccer_referee_supervisor.py", line 1, in <module>
    from controller import Supervisor
  File "C:\Program Files\Webots\lib\controller\python37\controller.py", line 31, in <module>
    import _controller
ImportError: DLL load failed: The specified module could not be found.
WARNING: 'rcj_soccer_referee_supervisor' controller exited with status: 1.

mrshu, 2020-12-14:
Hi @jamestalkington, thanks for the report!

@felipenmartins  this seems to be happening on Windows -- did you happen to run into something similar by any chance? I guess we'll be seeing more similar issues in the future.

felipenmartins, 2020-12-14:
Weird... I am running on Windows and all robots are moving.

mrshu, 2020-12-30:
@jamestalkington thank you for your patience here.

Could you please confirm that the other Webots demos work for you? Alternatively, could you please try to download the new version (Webots R2021a) and see if that helps?

Thanks!

ethanol20924, 2021-01-05:
Hi, I just quickly tested R2021a on my windows laptop and it does also does not work. Running on Windows 10 Build 19042. It appears to be an error with the reflog:

```Traceback (most recent call last):
  File "C:\Users\admin\OneDrive\Documents\Github\jr-soccer-simulation\controllers\rcj_soccer_referee_supervisor\rcj_soccer_referee_supervisor.py", line 48, in <module>
    if not referee.tick():
  File "C:\Users\admin\OneDrive\Documents\Github\jr-soccer-simulation\controllers\rcj_soccer_referee_supervisor\referee\referee.py", line 85, in tick
    self.log.event(
  File "C:\Users\admin\OneDrive\Documents\Github\jr-soccer-simulation\controllers\rcj_soccer_referee_supervisor\referee\json_logger.py", line 38, in event
    with self.logfile.open('a') as outfile:
  File "C:\Users\admin\AppData\Local\Programs\Python\Python39\lib\pathlib.py", line 1241, in open
    return io.open(self, mode, buffering, encoding, errors, newline,
  File "C:\Users\admin\AppData\Local\Programs\Python\Python39\lib\pathlib.py", line 1109, in _opener
    return self._accessor.open(self, flags, mode)
OSError: [Errno 22] Invalid argument: 'reflog\The_Blues_vs_The_Yellows-2021-01-05T12:19:39.766976Z.jsonl'
WARNING: 'rcj_soccer_referee_supervisor' controller exited with status: 1.```

jamestalkington, 2021-01-05:
@mrshu Thank you for your reply.

The other Webots demos work. I have tried downloading the new version (Webots R2021a) but the issue is not resolved.

I have also tried on my other computer that runs Windows 10 and have gotten a different issue that appears to be the same as @ethanol20924:

OS: Windows 10 Build 19042.685
Python: 3.8
Webots: R2021a

```INFO: rcj_soccer_player: Starting controller: python.exe -u rcj_soccer_player.py
INFO: rcj_soccer_player: Starting controller: python.exe -u rcj_soccer_player.py
INFO: rcj_soccer_player: Starting controller: python.exe -u rcj_soccer_player.py
INFO: rcj_soccer_player: Starting controller: python.exe -u rcj_soccer_player.py
INFO: rcj_soccer_player: Starting controller: python.exe -u rcj_soccer_player.py
INFO: rcj_soccer_player: Starting controller: python.exe -u rcj_soccer_player.py
INFO: rcj_soccer_referee_supervisor: Starting controller: python.exe -u rcj_soccer_referee_supervisor.py
Traceback (most recent call last):
  File "rcj_soccer_referee_supervisor.py", line 48, in <module>
    if not referee.tick():
  File "C:\Users\Admin\Documents\webots-soccer-sim-playground\controllers\rcj_soccer_referee_supervisor\referee\referee.py", line 85, in tick
    self.log.event(
  File "C:\Users\Admin\Documents\webots-soccer-sim-playground\controllers\rcj_soccer_referee_supervisor\referee\json_logger.py", line 38, in event
    with self.logfile.open('a') as outfile:
  File "C:\Users\Admin\AppData\Local\Programs\Python\Python38\lib\pathlib.py", line 1200, in open
    return io.open(self, mode, buffering, encoding, errors, newline,
  File "C:\Users\Admin\AppData\Local\Programs\Python\Python38\lib\pathlib.py", line 1054, in _opener
    return self._accessor.open(self, flags, mode)
OSError: [Errno 22] Invalid argument: 'reflog\\The_Blues_vs_The_Yellows-2021-01-05T11:45:34.317206Z.jsonl'
WARNING: 'rcj_soccer_referee_supervisor' controller exited with status: 1```

mrshu, 2021-01-05:
Hi @jamestalkington and @ethanol20924,

Thank you for trying this out!

I believe this is an issue we actually did run into (it turns out Windows does not support `:` as part of the filename) and it should be fixed in the current `master`.

Could you please download the latest updates from the `master` branch and see if the problem gets resolved?

Thanks!

ethanol20924, 2021-01-06:
Hi @mrshu, it seems to work for me now. Thanks so much fixing this!

jamestalkington, 2021-01-06:
Hi @mrshu 
Problem has been fixed for me on both of my Windows 10 devices. Thank you for taking the time to solve the issue.

closed 2021-01-06 by jamestalkington