robocup-junior/rcj-soccersim #29: Will inter-robot communication be supported?

mattyoung101, 2020-12-27:
Hello,
I'm just wondering if communication between robots will be supported in the simulation. We are planning to use this to better organise strategies between the robots. If it will be supported, will this be through something like Protobuf or Python's pickle serialiser?

Thanks.

mrshu, 2020-12-29:
Hi @mattyoung101,

Thanks for the question!

While I cannot really promise anything and need to note that we are currently trying to prepare a Minimum Viable Version (i.e. the simplest one possible) of the Simulator, inter-robot communication is something we definitely want the simulator to eventually support. Whether it will be ready for the January/February 2021 tournament I cannot really say, but I would not count on it. There is other information that you can probably already utilize to organize strategies between the robots on your team (each robot "sees" everything on the field fro instance).

That said, if it came to some communication being implemented, I believe we'd probably go the Protobuf way before pickling Python objects (the way I see it, there is a good chance we'd probably just send JSON objects around to make it relatively easy to debug). Something like that is already taking place -- see https://github.com/RoboCupJuniorTC/webots-soccer-sim-playground/blob/master/controllers/rcj_soccer_referee_supervisor/referee/supervisor.py#L138-L188

Thanks again for asking -- I hope my lukewarm answer will not deter you from putting together some nice robot programs together and having fun while doing so!

Should you have any more questions, please feel free to respond either to this thread or create a new issue.

Thanks!

cc @Adman

mrshu, 2020-12-29:
cc @aquahika @marcodankel @felipenmartins @javier687

mattyoung101, 2021-01-12:
Hi @mrshu, I was reading through the rules and noticed rule 3.4, "Communication", which says:

> Robots may communicate within each team in the scope of the simulation, as long as they abide by rule Rule 3.3, “Interference”.

Seeing this, I'd like to ask if it's acceptable to use inter-process communication to exchange data between our team's robots. The plan was to have robot 1 open a TCP socket on localhost with a programmatically assigned port, which robots 2 and 3 would then connect to and exchange info over.

The rule says "within the scope of the simulation" - I'm not sure if that means, for example, using a Webots receiver, or whether this setup would be acceptable. Thanks in advance.

mrshu, 2021-01-14:
Hi @mattyoung101,

First of all, let me tell you that we find the fact that you are asking questions like this very positive, for it is very close to what we would have done in your place :)

> The rule says "within the scope of the simulation" - I'm not sure if that means, for example, using a Webots receiver, or whether this setup would be acceptable.

The intention of the rule was indeed for the controllers to only make use of the APIs provided by Webots. The reason behind is actually very simple: Webots APIs are something we are willing and (it seems like) are capable of supporting. We cannot really guarantee that you will be able to open a TCP socket on one robot and connect to it from the other, so I'd strongly suggest you implement some failovers in case it fails. 

That said, we will also not explicitly forbid something like this, as we would forbid a ton of possible innovation as a byproduct. In my view, this byproduct is what we are really after anyway :slightly_smiling_face:.

Hope this helps!

Adman, 2021-03-20:
@mattyoung101 Hello, we have created a draft for inter-robot communication. See PR #90

Adman, 2021-09-06:
@mattyoung101 The communication feature is now live :)

closed 2021-09-06 by Adman

felipenmartins, 2021-09-07:
Top! Nice work! :-)

commit 31fd2305 (2021-03-20): Add inter-robot communication

* closes #29
commit 1553aa55 (2021-03-20): Add inter-robot communication

* closes #29
commit b4167872 (2021-03-20): Add inter-robot communication

* closes #29
commit 330557f3 (2021-03-20): Add inter-robot communication

* closes #29