robocup-junior/rcj-soccersim #31: Template for participants

Adman, 2020-12-29:
@mrshu @felipenmartins I was thinking that we could provide some template for participants. PR https://github.com/RoboCupJuniorTC/webots-soccer-sim-playground/pull/30/ contains demo programs, where we are using this template as well.

```python
###### REQUIRED in order to import abstract robot class
import sys
from pathlib import Path
sys.path.append(str(Path('.').absolute().parent))
from abstract_robot import AbstractRobot, TIME_STEP
######

import math

# Feel free to import other built-in libraries here
# You can import also other helper scripts that you create

class MyRobot(AbstractRobot):
    def run(self):
        while self.robot.step(TIME_STEP) != -1:
            if self.is_new_data():
                # READ NEW DATA FROM SUPERVISOR
                data = self.get_new_data()

                # WRITE YOUR CODE HERE

my_robot = MyRobot()
my_robot.run()
```

mrshu, 2020-12-29:
@Adman I believe the vast majority of participants will base their work off of the demo programs you've put together in #30. If they happen to contain a template like this, there is a good chance many of them will just include it in their programs.

That said, I think it makes a ton of sense and I would extend the comment a bit to say that the code shared among robots can be put into modules that are in the parent directory, much like `abstract_robot` in your example.

mrshu, 2021-01-04:
@Adman the files currently essentially look like this, right?

Adman, 2021-01-04:
@mrshu Yeah, like this. The only difference are robots B1 and Y1, which do not contain Path hacks.

closed 2021-02-08 by Adman