robocup-junior/rcj-soccersim #33: Change repository name

mrshu, 2020-12-30:
Hey @RoboCupJuniorTC/soccer-simulator,

I believe it is about time we've changed the name of the repository. Although it still resembles a playground, I think `rcj-soccer-simulator` or `rcj-soccer-sim` would probably make it a bit more descriptive than what we currently have.

Once we do that, we can also have a website on the order of `rcj-soccer-sim.github.io`, which could host the documentation as well as any other related information.

GitHub is handling renames [relatively well](https://docs.github.com/en/free-pro-team@latest/github/administering-a-repository/renaming-a-repository) (i.e. all the references to the past name will be forwarded to the new one), so I believe the change ought to be relatively painless.

What do you think? Would something like this make sense to you?

felipenmartins, 2020-12-31:
This makes sense to me. 
I vote for  rcj-soccer-sim .
:-)

mrshu, 2021-01-02:
Thanks everyone, I'll go ahead with `rcj-soccer-sim`

mrshu, 2021-01-02:
Resolved, thank you everyone!

closed 2021-01-02 by mrshu