robocup-junior/rcj-soccersim #35: Setup isort, black and flake8

Adman, 2020-12-30:
* Create requirements.txt with these packages (use pip-tools for this) + pre-commit package
* Setup pre-commit hook to check for code style
* Setup some CI (Jenkins, CircleCI or whatever else)

mrshu, 2021-01-02:
> Setup some CI (Jenkins, CircleCI or whatever else)

I guess we can try GitHub Actions, since we are here on GitHub already :)

> Add info about running code style checkers into README

I believe our long-term policy ought to be that the "simulator" ought not to have any outside dependencies (I don't even what to know what installing them on Windows would entail) but there are some packages that can be of help when it comes to its development.

Adman, 2021-01-02:
> I guess we can try GitHub Actions, since we are here on GitHub already :)

Of course, didn't know about this.

>  I believe our long-term policy ought to be that the "simulator" ought not to have any outside dependencies (I don't even what to know what installing them on Windows would entail) but there are some packages that can be of help when it comes to its development.

Yeah, simulator will be withou dependencies, those can be installed when contributing to the simulator itself / creating PR

mrshu, 2021-01-02:
@Adman Just FYI: I've already set up the GitHub actions for the docs (https://robocupjuniortc.github.io/rcj-soccer-sim/, see the config at https://github.com/RoboCupJuniorTC/rcj-soccer-sim/blob/master/.github/workflows/docs.yml), so I guess it should be relatively straightforward to set it in there.

commit 8097731e (2021-01-06): add isort, black and flake8 checkers

* create pre-commit hook config file
* add info to README
* create workflow build checkers
* closes #35
commit 8fb128d3 (2021-01-06): add isort, black and flake8 checkers

* create pre-commit hook config file
* add info to README
* create workflow build checkers
* closes #35
commit ecc5d775 (2021-12-01): add isort, black and flake8 checkers

* create pre-commit hook config file
* add info to README
* create workflow build checkers
* closes #35
PR #51 (2021-12-01): add isort, black and flake8 checkers
* create pre-commit hook config file
* add info to README
* create workflow build checkers
* closes #35
closed 2021-12-01 by Adman