robocup-junior/rcj-soccersim #36: Controller Error Run in MacOS

2black0, 2021-01-01:
This error happen while I run the Simulation in Mac OS

```
INFO: rcj_soccer_player: Starting controller: python -u rcj_soccer_player.py
INFO: rcj_soccer_player: Starting controller: python -u rcj_soccer_player.py
INFO: rcj_soccer_player: Starting controller: python -u rcj_soccer_player.py
INFO: rcj_soccer_player: Starting controller: python -u rcj_soccer_player.py
INFO: rcj_soccer_player: Starting controller: python -u rcj_soccer_player.py
INFO: rcj_soccer_player: Starting controller: python -u rcj_soccer_player.py
INFO: rcj_soccer_referee_supervisor: Starting controller: python -u rcj_soccer_referee_supervisor.py
  File "rcj_soccer_player.py", line 11
    def parse_supervisor_msg(packet: str) -> dict:
                                   ^
SyntaxError: invalid syntax
WARNING: 'rcj_soccer_player' controller exited with status: 1.
  File "rcj_soccer_player.py", line 11
    def parse_supervisor_msg(packet: str) -> dict:
                                   ^
SyntaxError: invalid syntax
WARNING: 'rcj_soccer_player' controller exited with status: 1.
  File "rcj_soccer_player.py", line 11
    def parse_supervisor_msg(packet: str) -> dict:
                                   ^
SyntaxError: invalid syntax
WARNING: 'rcj_soccer_player' controller exited with status: 1.
  File "rcj_soccer_player.py", line 11
    def parse_supervisor_msg(packet: str) -> dict:
                                   ^
SyntaxError: invalid syntax
WARNING: 'rcj_soccer_player' controller exited with status: 1.
  File "rcj_soccer_player.py", line 11
    def parse_supervisor_msg(packet: str) -> dict:
                                   ^
SyntaxError: invalid syntax
WARNING: 'rcj_soccer_player' controller exited with status: 1.
  File "rcj_soccer_player.py", line 11
    def parse_supervisor_msg(packet: str) -> dict:
                                   ^
SyntaxError: invalid syntax
WARNING: 'rcj_soccer_player' controller exited with status: 1.
  File "rcj_soccer_referee_supervisor.py", line 14
    directory: Path,
             ^
SyntaxError: invalid syntax
WARNING: 'rcj_soccer_referee_supervisor' controller exited with status: 1.
```

Adman, 2021-01-01:
Hello @2black0 . The problem is that you are using python version 2 for executing the controller. README now contains the way how to setup python3 https://github.com/RoboCupJuniorTC/webots-soccer-sim-playground#how-do-i-try-this-out.
Feel free to ask further if this won't work for you.

2black0, 2021-01-01:
thanks for your reply @Adman 
setup python3 command in preference was done

and I can confirm my python3 version is 3.8.5

[image]

but when start simulation, error still happen

```
INFO: rcj_soccer_player: Starting controller: python3 -u rcj_soccer_player.py
INFO: rcj_soccer_player: Starting controller: python3 -u rcj_soccer_player.py
INFO: rcj_soccer_player: Starting controller: python3 -u rcj_soccer_player.py
INFO: rcj_soccer_player: Starting controller: python3 -u rcj_soccer_player.py
INFO: rcj_soccer_player: Starting controller: python3 -u rcj_soccer_player.py
INFO: rcj_soccer_player: Starting controller: python3 -u rcj_soccer_player.py
INFO: rcj_soccer_referee_supervisor: Starting controller: python3 -u rcj_soccer_referee_supervisor.py
Traceback (most recent call last):
  File "rcj_soccer_player.py", line 2, in <module>
    from controller import Robot
  File "/Applications/Webots.app/lib/controller/python38/controller.py", line 31, in <module>
    import _controller
ImportError: dlopen(/Applications/Webots.app/lib/controller/python38/_controller.so, 2): Library not loaded: /Library/Frameworks/Python.framework/Versions/3.8/Python
  Referenced from: /Applications/Webots.app/lib/controller/python38/_controller.so
  Reason: image not found
WARNING: 'rcj_soccer_player' controller exited with status: 1.
Traceback (most recent call last):
  File "rcj_soccer_player.py", line 2, in <module>
    from controller import Robot
  File "/Applications/Webots.app/lib/controller/python38/controller.py", line 31, in <module>
    import _controller
ImportError: dlopen(/Applications/Webots.app/lib/controller/python38/_controller.so, 2): Library not loaded: /Library/Frameworks/Python.framework/Versions/3.8/Python
  Referenced from: /Applications/Webots.app/lib/controller/python38/_controller.so
  Reason: image not found
WARNING: 'rcj_soccer_player' controller exited with status: 1.
Traceback (most recent call last):
  File "rcj_soccer_player.py", line 2, in <module>
    from controller import Robot
  File "/Applications/Webots.app/lib/controller/python38/controller.py", line 31, in <module>
    import _controller
ImportError: dlopen(/Applications/Webots.app/lib/controller/python38/_controller.so, 2): Library not loaded: /Library/Frameworks/Python.framework/Versions/3.8/Python
  Referenced from: /Applications/Webots.app/lib/controller/python38/_controller.so
  Reason: image not found
WARNING: 'rcj_soccer_player' controller exited with status: 1.
Traceback (most recent call last):
  File "rcj_soccer_player.py", line 2, in <module>
    from controller import Robot
  File "/Applications/Webots.app/lib/controller/python38/controller.py", line 31, in <module>
    import _controller
ImportError: dlopen(/Applications/Webots.app/lib/controller/python38/_controller.so, 2): Library not loaded: /Library/Frameworks/Python.framework/Versions/3.8/Python
  Referenced from: /Applications/Webots.app/lib/controller/python38/_controller.so
  Reason: image not found
WARNING: 'rcj_soccer_player' controller exited with status: 1.
Traceback (most recent call last):
  File "rcj_soccer_player.py", line 2, in <module>
    from controller import Robot
  File "/Applications/Webots.app/lib/controller/python38/controller.py", line 31, in <module>
    import _controller
ImportError: dlopen(/Applications/Webots.app/lib/controller/python38/_controller.so, 2): Library not loaded: /Library/Frameworks/Python.framework/Versions/3.8/Python
  Referenced from: /Applications/Webots.app/lib/controller/python38/_controller.so
  Reason: image not found
WARNING: 'rcj_soccer_player' controller exited with status: 1.
Traceback (most recent call last):
  File "rcj_soccer_player.py", line 2, in <module>
    from controller import Robot
  File "/Applications/Webots.app/lib/controller/python38/controller.py", line 31, in <module>
    import _controller
ImportError: dlopen(/Applications/Webots.app/lib/controller/python38/_controller.so, 2): Library not loaded: /Library/Frameworks/Python.framework/Versions/3.8/Python
  Referenced from: /Applications/Webots.app/lib/controller/python38/_controller.so
  Reason: image not found
WARNING: 'rcj_soccer_player' controller exited with status: 1.
Traceback (most recent call last):
  File "rcj_soccer_referee_supervisor.py", line 6, in <module>
    from referee.referee import RCJSoccerReferee
  File "/Volumes/Storage/Download/webots-soccer-sim-playground-master/controllers/rcj_soccer_referee_supervisor/referee/referee.py", line 7, in <module>
    from referee.supervisor import RCJSoccerSupervisor
  File "/Volumes/Storage/Download/webots-soccer-sim-playground-master/controllers/rcj_soccer_referee_supervisor/referee/supervisor.py", line 5, in <module>
    from controller import Supervisor
  File "/Applications/Webots.app/lib/controller/python38/controller.py", line 31, in <module>
    import _controller
ImportError: dlopen(/Applications/Webots.app/lib/controller/python38/_controller.so, 2): Library not loaded: /Library/Frameworks/Python.framework/Versions/3.8/Python
  Referenced from: /Applications/Webots.app/lib/controller/python38/_controller.so
  Reason: image not found
WARNING: 'rcj_soccer_referee_supervisor' controller exited with status: 1.
```

2black0, 2021-01-02:
**[SOLVED]**

[image]

this is the way to solve this problem:
1. Install Python 3.8 via homebrew
`$ brew install python@3.8`
2. change python command in preference of webots to
`/usr/local/opt/python@3.8/bin/python3`

[image]

mrshu, 2021-01-02:
Thank you very much for your update @2black0!

We hope to soon have some documentation on how to set up the simulator, if you would agree, I think it would be great to add this in there (as for instance @marcodankel ran into a similar issue as well, if I am not mistaken).

I am closing this issue but please feel free to reopen it if I missed anything.

Thanks!

closed 2021-01-02 by mrshu

correll, 2022-01-21:
> 2\. /usr/local/opt/python@3.8/bin/python3

This unfortunately does not work for me. I still get "image not found", this time simply referencing the 3.8 directories vs. 3.9 that I have used before.

mrshu, 2022-01-21:
@correll would you mind pasting the full text of the error? We'll try to see what's up with that :slightly_smiling_face:

gueparducoin, 2022-04-19:
Hello, I have the same problem than correll, could you help me please ?

Here is the code : 

INFO: supervisor_controller: Starting controller: /usr/local/opt/python@3.8/bin/python3 -u supervisor_controller.py
INFO: ur10-controller: Starting controller: /usr/local/opt/python@3.8/bin/python3 -u ur10-controller.py
Traceback (most recent call last):
  File "supervisor_controller.py", line 3, in <module>
    from controller import Supervisor
  File "/Applications/Webots.app/lib/controller/python38/controller.py", line 31, in <module>
    import _controller
ImportError: dlopen(/Applications/Webots.app/lib/controller/python38/_controller.so, 2): Library not loaded: /Library/Frameworks/Python.framework/Versions/3.8/Python
  Referenced from: /Applications/Webots.app/lib/controller/python38/_controller.so
  Reason: image not found
WARNING: Se salió del controlador 'supervisor_controller' con estado: 1.
Traceback (most recent call last):
  File "ur10-controller.py", line 5, in <module>
    from controller import Robot, Keyboard
  File "/Applications/Webots.app/lib/controller/python38/controller.py", line 31, in <module>
    import _controller
ImportError: dlopen(/Applications/Webots.app/lib/controller/python38/_controller.so, 2): Library not loaded: /Library/Frameworks/Python.framework/Versions/3.8/Python
  Referenced from: /Applications/Webots.app/lib/controller/python38/_controller.so
  Reason: image not found
WARNING: Se salió del controlador 'ur10-controller' con estado: 1.

mrshu, 2022-04-19:
Thanks for reporting this @gueparducoin. From my first observations this seems to be connected with the way Webots is distributed.

@Adman, do you think this is something we should report upstream (i.e. to Webots) or is it something we deal with ourselves?

Adman, 2022-04-19:
@gueparducoin Have you tried fixing it the way @2black0 solved the issue? If yes and it still does not work, I would suggest contacting Webots team, since it might be problem in the installer for MacOS, like @mrshu says.