robocup-junior/rcj-soccersim #50: Deal with robots that enter infinite loop

Adman, 2021-01-05:
When a robot is not responding, end the match and the opponent wins 10:0.

mrshu, 2021-01-09:
How do you think we should go about doing this?

Here is what I could see as "doable but certainly not nice":
- Overload `get_new_data` and make it emit something each second (to a named pipe or to some file -- something like a heartbeat)
- Have an outside checker that will take a look at all the processes/files and within some grace period (say each 5 or 10 seconds) check whether the heartbeat came in
- If it did not, end the match

Do you think something like that could work?

Adman, 2021-01-10:
Yes, this is probably the best doable solution at the moment. We should probably mention this in the rules - let's discuss it in the meeting.

mrshu, 2021-03-16:
@Adman Guess what, it's got a very simple solution in the end!

Seems like the home-take point is that we should have just asked...

https://github.com/cyberbotics/webots/issues/2859#issuecomment-800042125

commit ecf7c002 (2021-03-16): world: Desynchronize the Robots

* Previously, infinite loops resulted in halting the whole Webots
  application.

* As suggested in
  https://github.com/cyberbotics/webots/issues/2859#issuecomment-800042125
  setting synchronize synchronization to `FALSE` on all the robots ought
  to alleviate this problem.

* Fixes #50.
PR #86 (2021-03-16): world: Desynchronize the Robots
* Previously, infinite loops resulted in halting the whole Webots
  application.

* As suggested in
  https://github.com/cyberbotics/webots/issues/2859#issuecomment-800042125
  setting synchronize synchronization to `FALSE` on all the robots ought
  to alleviate this problem.

* Fixes #50.
commit 1b581da2 (2021-03-16): world: Desynchronize the Robots (#86)

* Previously, infinite loops resulted in halting the whole Webots
  application.

* As suggested in
  https://github.com/cyberbotics/webots/issues/2859#issuecomment-800042125
  setting synchronize synchronization to `FALSE` on all the robots ought
  to alleviate this problem.

* Fixes #50.
closed 2021-03-16 by mrshu