robocup-junior/rcj-soccersim #55: Wheels slowly drifting away from the robot

Bluskript, 2021-01-13:
[image]

I don't know what has been causing this, it might be a bug related to rotation since I tweaked those values. I'll report any findings related to this bug if I find any.

Bluskript, 2021-01-16:
found minimal reproducible test case:
```diff
# rcj_soccer_player controller - ROBOT Y3

###### REQUIRED in order to import files from Y1 controller
import sys
from pathlib import Path

sys.path.append(str(Path(".").absolute().parent))
# You can now import scripts that you put into the folder with your
# robot B1 controller
from rcj_soccer_player_y1 import rcj_soccer_robot, utils

######

# Feel free to import built-in libraries
import math

class MyRobot(rcj_soccer_robot.RCJSoccerRobot):
    def run(self):
        while self.robot.step(rcj_soccer_robot.TIME_STEP) != -1:
            if self.is_new_data():
                data = self.get_new_data()

                # Get the position of our robot
                robot_pos = data[self.name]
                # Get the position of the ball
                ball_pos = data["ball"]

                # Get angle between the robot and the ball
                # and between the robot and the north
                ball_angle, robot_angle = self.get_angles(ball_pos, robot_pos)

                # Compute the speed for motors
-                 direction = utils.get_direction(ball_angle)
+                direction = 1

                # If the robot has the ball right in front of it, go forward,
                # rotate otherwise
                if direction == 0:
                    left_speed = -5
                    right_speed = -5
                else:
                    left_speed = direction * 4
                    right_speed = direction * -4

                # Set the speed to motors
                self.left_motor.setVelocity(left_speed)
                self.right_motor.setVelocity(right_speed)

my_robot = MyRobot()
my_robot.run()
```

Adman, 2021-02-21:
Hello @Bluskript. Could you please test if this problem occurs with the newest software?

I'll close this issue for now. If you ran into similar problem like you reported, feel free to reopen this issue.

closed 2021-02-21 by Adman