robocup-junior/rcj-soccersim #60: Need a quick way to determine which side you're on

TBSDrJ, 2021-01-21:
I have just been looking at the code for a couple of days (I am mentoring a couple of teams, we registered one this morning, and I'll have at least two more tomorrow).  Thank you to all who have put so much work into it! 

I see an issue you might have when you run the submitted code for the competition.  Teams will be assigned as Yellow or Blue for each match, so you'll be substituting the three folders for the three robots in the simulation.

The default import statement on line 9 of b2, b3, y2 and y3 needs to be changed depending on which side you're on, or else, if you load up a submitted team as Yellow, but the team has the b1 import in it, it'll import the _other_ team's `rcj_soccer_robot` and `utils` files instead of their own.

I would say to just force the teams to detect whether they are yellow or blue based on the orientation, but the data is not available yet at the import stage.  It can't go in the rcj_soccer_robot file because that import is the issue, it has to go in each of the three of the robot scripts.

The best suggestions I have are: 
1. Require each team to put `team = "BLUE"` or `team = "YELLOW"` on line 1 of all three robot's control code, and the human referees running the competition would need to set these correctly in each of the six robots' code at the beginning of each match.
2. Require each team to set a variable in their `rcj_soccer_robot` file, for ex. `TEAM_NAME = "Bishop's Knights A"` and then change the import in b2, b3, y2, y3 to something like:
```
from rcj_soccer_player_b1 import rcj_soccer_robot, utils
if rcj_soccer_robot.TEAM_NAME == "Bishop's Knights A":
    team = "BLUE"
else:
    from rcj_soccer_player_y1 import rcj_soccer_robot, utils
    team = "YELLOW"
```
This would make the correct import automatic, but requires teams to set this up correctly.  If the base code here had "Default Blue Team" and "Default Yellow Team" pre-set, that would be pretty easy.
3. Require each team to change the folder names to include their own team name, and reset the import statements to match.  Again, it requires the teams to set this up correctly, and it also requires some change to the import code.  I tried changing the folder name, and the folder wasn't found.

I feel best about #2, if the base code has this set up with default team names, it seems both most likely to succeed and easiest for the people running the competition.  If you want me to set that (or something else) up, I can take a fork and submit a pull request.

Best,
Marcus Jaiclin
The Bishop's School
La Jolla, CA, USA

Adman, 2021-01-21:
Hello,
thanks for raising this question.
We have been thinking about this and our decision is that we do not want to be changing the code/import statements/whatever the participants submit.
The way we are going to deal with this is that each team should put their name or some unique identifier into the name of the controller folders. Based on this, the world file will be generated using our script https://github.com/RoboCupJuniorTC/rcj-soccer-sim/tree/master/scripts, so it will be possible to import stuff from other controller folder since you know the name.

We will provide more information soon.

If you have any further questions, I'm glad to answer them.

Adrian

cc @mrshu @felipenmartins @aquahika

closed 2021-01-21 by TBSDrJ

Adman, 2021-02-03:
Hello,

so to answer the question about determining the name of the team, we described it in https://robocupjuniortc.github.io/rcj-soccer-sim/how_to/.
You can just simply call `robot.getName()`, which returns one of the following strings `["B1", "B2", "B3", "Y1", "Y2", "Y3"]` and therefore based on first character, it's easy to check which team you are in :)

Hope it helps!