robocup-junior/rcj-soccersim #71: soccer.wbt.template robot names

TBSDrJ, 2021-02-06:
I tried to set up a new simulation world using in scripts > README.md.

In the template world in scripts > templates > soccer.wbt.template, the robots are named "b1", "b2", etc.

In worlds > soccer.wbt, the robots are named "B1", "B2", etc. 

This was a breaking change for me (self.name evaluated to "b1" but data uses key "B1"), and worked when I capitalized them.  I wasn't sure if this was somehow related to my setup, so I didn't submit a pull request for it.

If this is not just me, the change needed would be all in the file scripts > templates > soccer.wbt.template on lines: 511, 638, 762, 886, 1010, and 1134.  Each change is just "b1" to "B1", etc.

mrshu, 2021-02-06:
Thanks for the bug report @TBSDrJ, this is indeed a bug and I'll do my best to fix it shortly.

TBSDrJ, 2021-02-06:
I'll fork it and post a pull request so you can just merge it.

mrshu, 2021-02-08:
@TBSDrJ thanks again for the fix, I believe we can close this.

closed 2021-02-08 by mrshu