robocup-junior/rcj-soccersim #98: Detecting ball lack of progress when the ball is actually moving

alibahmanyar, 2021-05-02:
Hello,
Occasionally, A "ball: lack of progress" is detected and the ball is moved to the center point, after that it takes some time for the robots to reach the ball, but when they reach it and kick it, after several seconds another lack of progress is detected even though the ball is actually moving significantly at that time. However, I'm not sure if this is a bug or if it's actually supposed to be like this (because the total movement of the ball in a certain time window is actually less than a certain threshold).
An example is attached in which the situation occurs around 107 seconds after the initial kickoff. There is also an mp4 clip demonstrating the case:
[test.zip](https://github.com/RoboCupJuniorTC/rcj-soccer-sim/files/6411628/test.zip)

Thanks

mrshu, 2021-05-17:
Thanks for the report @alibahmanyar!

@Adman would you mind taking a quick look?

Adman, 2021-05-17:
I have no exact idea how this could happen. Could be some edge case where the lack-of-progress counter is almost done, robot pushes the ball and then the referee calls lack-of-progress because the ball did not move enough.

We will keep this in mind but I think it's not a big problem currently.

mrshu, 2021-12-22:
@alibahmanyar @Adman there hasn't been much happening here for the past 180 days -- let me close the issue for now but do not be afraid to comment on it and/or re-open it in the future!

closed 2021-12-22 by mrshu