robocup-junior/rcj-soccersim #127: the simulation doesn't work

mohammadhosseinz, 2022-01-22:
The page simulator is parallel with gravity
It seems that the land has 90 degrees of error, in this "73bb7e534982e4f71f23637d832837fb92d655e0" commit.

Adman, 2022-01-22:
@mohammadhosseinz Hello, what version of webots do you have? Could you please paste a screenshoot which describes you problem?

mohammadhosseinz, 2022-01-23:
Hi, the last version. 189 is the commit number.
[image]
Thanks for your attention.

Adman, 2022-01-23:
Alright, so you have the newest version of rcj-soccer-sim.

But how about Webots version? Is it R2021b, R2022a or something else?

mohammadhosseinz, 2022-01-25:
we tested in macOS and windows in windows did not work in both version R2022a and R2021b but in mac is ok.

Adman, 2022-01-25:
Do you have Windows 11? I tried running it on both Win11 and Win10 and it works.

mohammadhosseinz, 2022-01-26:
tested on windows 7

Adman, 2022-01-31:
Not sure what might have caused the problem. Do you have Win10/11 available?

mohammadhosseinz, 2022-02-21:
problem solved😄
I noticed this problem when I run version 2 with webots version 2021 b
thank you for your attention🙏

mrshu, 2022-02-21:
Thanks for getting back to us @mohammadhosseinz !

closed 2022-02-21 by mrshu