robocup-junior/rcj-soccersim #130: Robot relocation bug

RichoM, 2022-04-17:
Hi, I've been testing the simulator a lot lately and I've noticed a problem with the relocation of robots. A lot of times when the supervisor relocates a robot, it ends up flipped upside down (or to the side, in the worst case).

To show what I'm talking about I recorded a short video. These are all instances of robots being flipped (and a few close calls) during just one match: 

https://youtu.be/4vt4nLYVCsI

- Webots version: r2022a
- OS: macOs Mojave version 10.14.6
- Simulator: Latest (2278b6d48b612231d27884aa013389be6c5a84c9)

I'll see if I can test in a windows computer and I'll let you know if the problem occurs there as well.

Adman, 2022-04-17:
Hello @RichoM, thanks for reporting the problem.

The code responsible for moving the object is located [here](https://github.com/RoboCupJuniorTC/rcj-soccer-sim/blob/master/controllers/rcj_soccer_referee_supervisor/referee/supervisor.py#L77-L87)

```python
tr_field = self.robot_translation_fields[robot_name]
tr_field.setSFVec3f(position)
self.robot_nodes[robot_name].resetPhysics()
```

It might be possible that the robot is running at high speed while being relocated. Since I'm not able to reproduce the problem so often, could you please change the order of these two lines in supervisor's code and test it? Thanks!
```python
tr_field.setSFVec3f(position)
self.robot_nodes[robot_name].resetPhysics()
```

RichoM, 2022-04-17:
Thanks for the quick response. I tested your proposed fix but it didn’t seem to have any effect.

I think you’re right in that it could be related to the robot speeds so I tried to change the team code to detect lack of progress and stop the motors a few moments before being relocated. That seemed to help a little but not as much as I hoped.

To help reproduce the problem more easily I made the following team controller:

```python
from controller import Robot

TIME_STEP = 32

robot = Robot()

left_motor = robot.getDevice("left wheel motor")
right_motor = robot.getDevice("right wheel motor")

left_motor.setPosition(float('+inf'))
right_motor.setPosition(float('+inf'))

while robot.step(TIME_STEP) != -1:
    left_motor.setVelocity(10.0)
    right_motor.setVelocity(10.0)
```

With the code above running for both teams I can reproduce the bug in almost 100% of relocations (at least on my computer).

RichoM, 2022-04-18:
I've been doing some more testing and I think the problem is related to this issue: https://github.com/cyberbotics/webots/issues/2899

So I added a similar workaround for the robots and it seems to help a lot (see https://github.com/GIRA/rcj-soccer-sim/commit/5904d008e20035dc7e92f63cd1c35353e235a970). It still happens ocassionally but it's *much* less frequently than before.

If you want I can make a pull request with this commit.

Adman, 2022-04-19:
Thanks, I tested your example controller and robots indeed flip over quite often. When I decreased the velocity to just 5, it does not happen, but I can see robots being "shaked" (almost flipped over). I'll ask cyberbotics guys whether this could be related to the issue you linked.

Adman, 2022-04-22:
@RichoM Could you please try to set `OBJECT_DEPTH` to 0.0375 according to this comment https://github.com/cyberbotics/webots/issues/2899#issuecomment-1106612329 ?

RichoM, 2022-04-22:
I tried setting `OBJECT_DEPTH` to 0.0375 and it didn't work, the robots keep falling. 

I also tested with a bunch of different values just to be sure, none of them worked either.

```python
OBJECT_DEPTH = 0.042
#OBJECT_DEPTH = 0.0375
#OBJECT_DEPTH = 0.0373044 
#OBJECT_DEPTH = 0.0375232
#OBJECT_DEPTH = 0.0
#OBJECT_DEPTH = 0.021
#OBJECT_DEPTH = 0.0315
#OBJECT_DEPTH = 0.03675
#OBJECT_DEPTH = 0.05
#OBJECT_DEPTH = 0.1
#OBJECT_DEPTH = 0.027
#OBJECT_DEPTH = 0.03817
#OBJECT_DEPTH = 0.075
```

Adman, 2022-04-25:
@RichoM Thanks. There's one more comment https://github.com/cyberbotics/webots/issues/2899#issuecomment-1108128674 to  test the object depth with nightly build / source build. Could you please give it a try?

RichoM, 2022-04-25:
I've just tested the nightly build with `OBJECT_DEPTH = 0.0375` and it definitely works better. However, something else seems to be broken.

I've recorded a new short video. Look at the two blue robots pushing each other inside the yellow goal. The one that gets flipped over is not the one that is relocated.

https://user-images.githubusercontent.com/1226503/165183378-6c45bda1-4f4b-4737-9643-424588da2656.mov

Adman, 2022-04-26:
@RichoM Thanks. I watched it and I think the problem is that when the robot B3 is being relocated (the one deeper inside the yellow goal), the collision detection might not be completely turned off for this robot and it pushes the other robot resulting in flipover.

RichoM, 2022-04-26:
Yes, I just confirmed that the same happens using the latest webots release after I applied the `resetPhysics()` workaround to the supervisor. I think I never noticed it before because having two robots pushing against the wall at max speed like that is not a very realistic test.

What do you think we should do with the relocation bug? Wait for the next webots version or should we apply the workaround and have it fixed now?
I've been testing extensively with more realistic controllers and since I added the workaround I never saw a robot being flipped over because of a relocation.

Adman, 2022-04-28:
@RichoM Let's apply the workaround. Could you please create a pull request?

Adman, 2022-04-30:
@Najmehsarfi It's in consts.py in referee's code.

closed 2022-04-30 by Adman