robocup-junior/rcj-soccersim #131: Controller error on MacOS

gueparducoin, 2022-04-19:
Hello, I'm currently using WEBOTS and here is my issue : 

INFO: supervisor_controller: Starting controller: /usr/local/opt/python@3.8/bin/python3 -u supervisor_controller.py
INFO: ur10-controller: Starting controller: /usr/local/opt/python@3.8/bin/python3 -u ur10-controller.py
Traceback (most recent call last):
  File "supervisor_controller.py", line 3, in <module>
    from controller import Supervisor
  File "/Applications/Webots.app/lib/controller/python38/controller.py", line 31, in <module>
    import _controller
ImportError: dlopen(/Applications/Webots.app/lib/controller/python38/_controller.so, 2): Library not loaded: /Library/Frameworks/Python.framework/Versions/3.8/Python
  Referenced from: /Applications/Webots.app/lib/controller/python38/_controller.so
  Reason: image not found
WARNING: Se salió del controlador 'supervisor_controller' con estado: 1.
Traceback (most recent call last):
  File "ur10-controller.py", line 5, in <module>
    from controller import Robot, Keyboard
  File "/Applications/Webots.app/lib/controller/python38/controller.py", line 31, in <module>
    import _controller
ImportError: dlopen(/Applications/Webots.app/lib/controller/python38/_controller.so, 2): Library not loaded: /Library/Frameworks/Python.framework/Versions/3.8/Python
  Referenced from: /Applications/Webots.app/lib/controller/python38/_controller.so
  Reason: image not found
WARNING: Se salió del controlador 'ur10-controller' con estado: 1.

mrshu, 2022-06-04:
Hi @gueparducoin, is this still a problem for you? Are you by any chance using the M1 version of macOS?

Thanks!

Adman, 2022-09-21:
I'll close the issue for now. If you need help just reopen it.

closed 2022-09-21 by Adman